instantlinux/docker-tools #217: "Duplicate driver instance detected ..." on Debian Trixie

Ivano62, 2025-08-22:
Thank you for the Software.
I have a problem with  instantlinux/nut-upsd on Debian Trixie and Docker version 28.3.3, build 980b856, the container doesn't start and the logs show (repeatedly):

_...$ docker container logs nut-upsd -f
Network UPS Tools - UPS driver controller 2.8.2
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
Startup timer elapsed, continuing...
Using protocol: Megatec/old 0.08
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
Duplicate driver instance detected (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists)! Terminating other driver!
Duplicate driver instance detected (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists)! Terminating other driver!
Can't initialise data from the UPS
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
USB communication driver (libusb 1.0) 0.47
Startup timer elapsed, continuing...
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
Duplicate driver instance detected (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists)! Terminating other driver!
Duplicate driver instance detected (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists)! Terminating other driver!
Duplicate driver instance detected (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists)! Terminating other driver!
Using protocol: Megatec/old 0.08
Duplicate driver instance is still alive (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists) after several termination attempts! Killing other driver!
Duplicate driver instance is still alive (could signal the process)
Startup timer elapsed, continuing...
Driver [tecnoware_1100] PID 49 initially exceeded maxstartdelay and is still starting
Network UPS Tools - UPS driver controller 2.8.2
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
Startup timer elapsed, continuing...
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
Duplicate driver instance detected (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists)! Terminating other driver!
Using protocol: Megatec/old 0.08
Duplicate driver instance detected (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists)! Terminating other driver!
Duplicate driver instance detected (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists)! Terminating other driver!
Can't initialise data from the UPS
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
USB communication driver (libusb 1.0) 0.47
Startup timer elapsed, continuing...
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
Duplicate driver instance detected (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists)! Terminating other driver!
Duplicate driver instance detected (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists)! Terminating other driver!
Duplicate driver instance detected (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists)! Terminating other driver!
Using protocol: Megatec/old 0.08
Duplicate driver instance is still alive (PID file /var/run/nut/nutdrv_qx-tecnoware_1100.pid exists) after several termination attempts! Killing other driver!
Duplicate driver instance is still alive (could signal the process)
Startup timer elapsed, continuing...
Driver [tecnoware_1100] PID 42 initially exceeded maxstartdelay and is still starting
Network UPS Tools - UPS driver controller 2.8.2
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
..._

The docker compose configuration is:

  _nut-upsd:
    image: instantlinux/nut-upsd
    container_name: nut-upsd
    environment:
      - TZ=...
      - DRIVER=nutdrv_qx
      - NAME=tecnoware_1100
      - DESCRIPTION=UPS Tecnoware 1100
      - PROTOCOL=Voltronic-QS
    devices:
      - /dev/bus/usb/001/002:/dev/bus/usb/001/002
    ports:
      - 3493:3493
    secrets:
      - nut-upsd-password
    restart: unless-stopped_

The same configuration works on Debian Bookworm.

Thank you in advance for your attention.

instantlinux, 2025-09-04:
Please try the `2.8.3-r2` tag and see if this still occurs. The upstream changelog has a lot of updates so this may have already been resolved; let me know if not.

instantlinux, 2025-09-04:
Glad to hear!

closed 2025-09-04 by instantlinux